rdugan/iceriver-oc #4: possible to make for ks3m?

harkijhoj, 2024-01-05:
i know that y be that cannot overclock. But t will be great if we can at least use your undervolt option to save them from heat and power bills. thanks a lot.

rdugan, 2024-01-06:
i do have a firmware for KS3M, it's in testing now and will be released soon.  If you find me in the Kaspa discord (pbfarmer) i can give you a pre-release copy.

However, it only supports clocks (no voltage yet), and as you suggested, overclocking potential on a KS3M is extremely limited.  A couple percent increase is probably all that's possible, and without a meter, i wouldn't recommend even attempting overclocking.

There is of course, the possibility of underclocking, but probably not too many people are interested in that.

harkijhoj, 2024-01-06:
Thank you. Would love to test it with voltage.or undervolt it. keep on
the great work.

KS3Muser, 2024-07-09:
How are you looking for the update on overclocking the latest KS3M?

rdugan, 2024-07-10:
> How are you looking for the update on overclocking the latest KS3M?

I'm not sure I understand your question.  KS3M hasn't changed since it was first released.  The KS3M firmware I've published should work on any KS3M.

nhhuayt, 2025-01-26:
I'm waiting for KS3M firmware that support voltage. I'm underclocking it for better efficiency.