rdugan/iceriver-oc #5: Problem with negative temperature

MrKTOTO, 2024-01-05:
when the temperature of the boards or chips is negative, the interface almost completely freezes and writes a notification that the machine is supposedly offline

rdugan, 2024-01-06:
I have never seen a negative temperature, and that would seem to indicate something is wrong with your asic, or at least the temperature probes.  But i will take a look to see if there's any reason the UI would not be able to deal with a negative temperature.

MrKTOTO, 2024-01-06:
nothing wrong with ASIC.  It in a container outside, and the temperature outside is -30 degrees).  So it turns out that the chips and boards cool down to about -10 while the ASIC is rebooting

MrKTOTO, 2024-01-06:
[image]

There is very cold outside)

chad5150, 2024-01-06:
go back to stock and see if the issue resolves. if so then the issues lies in the firmware possibly or like rdugan said about the temp probes. I wonder if extreme cold isnt having a negative effect also.

MrKTOTO, 2024-01-07:
This is the point: everything works fine on the stock firmware, but problems begin on custom ones. Most likely there are some errors in the fan speed calculation functions.  Or in the temperature display itself, I don’t know exactly, but in warm weather it works fine, but in cold weather the interface freezes

rdugan, 2024-01-07:
Are you saying you've had the machine in below freezing temperatures with stock firmware, and it was fine?  Because nothing i've done changes temperature readings (other than renaming them) or fan speed readings.  Also, generally when there is  a delay in the UI, its because the ASIC isn't responding, which wouldn't be surprising when it's freezing

chad5150, 2024-01-07:
Yeah -7 c translates to ~19 f. I agree with @rdugan and bring it in before it freezes.  Check out the machine specs on this page for optimal temp range. https://www.iceriver.io/product/iceriver-ks1/
Id bring it in just for testing purposes to see if the custom firmware interface still freezes. What are your clock offset and voltage offset values? What is your fan speed? Report back

MrKTOTO, 2024-01-08:
it doesn't matter, I noticed that when "Intake temp" has a negative value, the interface writes "Machine is offline" and the main page does not load, but when the ASIC warms up and the value becomes positive, then everything works fine

MrKTOTO, 2024-01-08:
on the original firmware, everything works with negative values, but there’s a glitch here

MrKTOTO, 2024-01-08:
I am 100% sure of what I say, everything was tested for several hours.

MrKTOTO, 2024-01-08:
the settings are applied, the mining setting page is loaded, but the main page is not.

rdugan, 2024-01-08:
The main page makes multiple requests to the hashboards/controller.  If they're not responding because they're frozen, then the page will not respond.

The main difference between mine and stock is that I show chip temperatures.  What is likely happening is that the boards/controller are unable to get the temperatures off frozen chips.

I can simulate responding with negative temperatures to check the UI, but I'm fairly certain it has nothing at all to do with the UI

rdugan, 2024-01-08:
Ok, as I expected, it is not a UI problem.  However, there is somewhere in my code that expects the temperatures to be unsigned.  I've made a change that I think will address this issue and have a new build.  Find me in the Kaspa discord if you want to test a pre-release version.  Otherwise, it will be in the next release (not sure on timing).

closed 2024-01-29 by rdugan