rdugan/iceriver-oc #15: password error, low hashrate

mndesign, 2024-01-24:
Suddently after 24h-48h the miner reboots and lands on a hashrate of about 94GH
When i try to log in to the miner i get password error.
I then have to reset the miner to get back in

rdugan, 2024-01-24:
all sorts of people are seeing this behavior, with or without my firmware, so i don't think it's related.  but i can possibly help investigate the issue if you find me in the Kaspa Discord.

mndesign, 2024-01-27:
sent you a friend request on discord since i cant message you :)
The issue has now evolved, and i cant even get into my KS0 after a factory reset.
I'm only getting the password error issue.
And the hashrate is between 50GH and 108GH

rdugan, 2024-01-29:
Password errors seem to be from file system corruption, as the miner can't read the password from the appropriate file.  May be fixed by reflashing/resetting to stock so the FS is refreshed.  Closing for now, but will update if anything more concrete is found.

closed 2024-01-29 by rdugan