rdugan/iceriver-oc #16: Ks3m stopped mining after ~30 minutes

dsmforce, 2024-01-29:
Hello!
I have a problem with my KS3M.
Ks3m stopped mining after ~30 minutes and i see very high temperature in some chips on hashboard 2,3 **but chips is COLD**.
i think that it's fake error. When i reboot the asic he worked good about 30 minutes and error again.

1. can you do in new firmware an automatic reboot when ks3m stopped mining?

[image]
[image]

rdugan, 2024-01-29:
I will try to look into possibilities here, but a simple automatic reboot during such conditions could be extremely dangerous.  If the ASIC is reporting high temps or any hardware fault for that matter, i have no way of distinguishing between real problems and garbage data.  In fact, we can't even be sure this isn't a cause for concern in your case.

demonbond, 2024-01-30:
My friend also had this problem, the firmware from the official website helped, the support also gave me a test firmware (I can send it to you), but I haven’t tried it.

dsmforce, 2024-01-31:
I’m swapped the first hashboard to second place and second to first place, and give the same error on 2(ex 1),3 hashboards.
I think that my control board works abnormally

dsmforce, 2024-02-08:
[image]

New error with new control board :(

rdugan, 2024-02-08:
Have you tried IceRivers new firmware?  I have a build that includes it, but it seems to have it's own instabilities (was causing some odd fan behavior on a normally working KS3M that I tested on)

dsmforce, 2024-02-08:
Yes, I tried all iceriver firmwares. 
Now I'm using a fan simulator and it doesn't matter

rdugan, 2024-02-09:
So i guess that means it didn't help?

dsmforce, 2024-02-09:
Yes