rdugan/iceriver-oc #18: Times on x-axis

jjphotos-jeremy, 2024-01-30:
There is something not quite right with the times on the new graphs. See attached screen shot.

This was taken at 16:18 (local time) with the refresh time at 16:15 (in the top left of screen). Both these times are correct.

The time on the hash rate graph is 2 hours old (which may be correct because of the 2 hour graph, but I'm not sure). The other graphs all show beyond the time of 16:15 and indeed the pop-up hint shows a time of 16:26, which is defiantly wrong as it's in the future, even as I type this!
[image]

Love the new graphs though!

rdugan, 2024-01-31:
Yep, its because i added more points than were originally displayed, and forgot to adjust the time scale calculation in the javascript.  Already fixed and ready for next release.  Thanks!

rdugan, 2024-02-10:
addressed in 0.7.0

closed 2024-02-10 by rdugan