rdugan/iceriver-oc #24: KS0 Pro stuck in reboot loop. Webgui comes online for few seconds, then miner restarts.

Deadly-Medley, 2024-02-05:
Installed the 0.6.0 firmware on my PS0 Pro. First time restart it came up fine, I clicked on the 2 confirmation boxes. I input my pool settings and saved. The miner did not start hashing, so I decided to restart. At that point it started the restart looping. I have tried factory reset via the button on front of the miner, but still no joy. At the moment it doesn't stay up long enough to try and restore factory firmware.

Deadly-Medley, 2024-02-05:
Further update. 
2nd Factory reset, the pool settings were clear again. The miner remain up, but not working. As soon as I entered the settings for my node and pool and saved, we resumed with the reboot loop behaviour.

After the 3rd factory reset via the front button on the device, it finally went back to factory IR firmware and I was able to get the miner up and running again.

rdugan, 2024-02-05:
This is certainly unusual behavior.  If you want to try to debug this - can you find me in the Kaspa Discord?

closed 2024-02-05 by rdugan

Deadly-Medley, 2024-02-06:
Just a quick update for those that may be reading this. I tried the same fimware upgrade again and this time it has worked flawlessly. It seems I was just unlucky first time round.