rdugan/iceriver-oc #26: KS0 Pro can't start mining on oc firmware

gike12, 2024-02-14:
Hi there,
I just got my ks0 pro a few days ago and when I install the firmware the mining does not start. On the official iceriver firmware it works but on the oc it seem not. On the 0.6.0 it connects to the pool but the Clock stays on 0.00 Mhz and loads but on the 0.7.0 the miner cannot connect to the pool and the Clock stays on 0.00 Mhz. I attach some screenshots maybe that can help.

on 0.6.0
[image]
[image]

on 0.7.0
[image]
[image]

And i don't know if its intended but on startup the webgui always says this:
[image]

rdugan, 2024-02-15:
0.6.0 pool status was unreliable.  this was addressed in 0.7.0.  the status indicators are explained here:
https://github.com/rdugan/iceriver-oc#uptime-and-job-rate-on-pool-status

bottom line, the asic is unable to connect to the pool...  this is an issue w/ your personal network (or less likely, the pool).  Maybe start here:
https://github.com/rdugan/iceriver-oc#known-issues

gike12, 2024-02-15:
Yes that was my suspicion as well. I had problems with my isp-s router for a long time...  I just ordered a new one and hopefully that will fix it. At that time I use the official firmware.

closed 2024-02-19 by gike12

rdugan, 2024-02-19:
@gike12 curious... did the new router fix the problem?

gike12, 2024-02-19:
Yeah it actually did. Just plugged into the new router upgraded the firmware and it instantly connected to the pool. Idk what changed with the new router but I'm happy it worked.