rdugan/iceriver-oc #27: Per Chip Clock & Voltage Control

BahmaNvidia, 2024-02-16:
Hi

thank you for your good work

do you have any plans on adding per chip clock and voltage control to the firmware?

it will be awesome to have individual control over the chips to maximize the performance within the temperature headroom of every chip

is it even possible to have per chip control?

rdugan, 2024-02-16:
Individual voltage control is not possible, though it may be possible at least for groups of chips on non-KS3* models.

Individual clock control may be possible, but this will take a lot of work and testing, and if I release something like this, it will likely be part of a more enterprise/farm focused firmware with a slightly higher fee.

BahmaNvidia, 2024-02-17:
That's great news!
 
Another issue I am having with my KS1 is that always one exact chip (number 13 in board 2) hangs about 10 minutes upon turning the machine on and temperature and voltage values of that chip freeze to the last value before hanging (even if i change the frequency and voltage it remains freezed to the last value)  and never change until i put the ASIC mode to "sleep" and back to "normal" after couple of minutes! then all the chips perform normally and change clock and voltage accordingly.

and that chip never hangs again until i restart the machine. 

if there is a hardware problem in that chip why won't it hang after putting the machine to sleep and back to normal? that's really confusing. 

and the annoying thing is putting the machine to sleep and back to normal makes it begin all the way from default frequency of 875 MHz and it takes a lot of time to reach the target again. can you please make it instant or at least increase in bigger steps after waking from sleep?

BoobFrog, 2024-02-20:
> > Individual voltage control is not possible, though it may be possible at least for groups of chips on non-KS3* models.
> > Individual clock control may be possible, but this will take a lot of work and testing, and if I release something like this, it will likely be part of a more enterprise/farm focused firmware with a slightly higher fee.
> 
> what about per board clock and voltage control?

That would be good per board, I think it would be enough since I have a board that I don't want to...

Thanks a lot for your work so far!

rdugan, 2024-02-21:
Nothing changes between boards and chips from what i stated above.  If voltage is controllable at all on KS3* machines, it appears it is only at the PSU.  clocks may be controllable per board.