rdugan/iceriver-oc #31: Clock and voltage differbetween boards

Jakkal, 2024-02-27:
Hi, and thank you for your great work!

I uploaded the new firmware to my Ks1 and Ks2. 
I noticed on the Ks1, that the clock speed and voltage on board number 2, differ from board 1 (and the overclock settings). I also noticed this in your photo here on GitHub. The difference changed, from slightly higher clock speed (1136.39mhz vs 1125mhz on board 1) and voltage (476mV vs 476,3mV on board 1) to have board 2 underperform in both voltage and clock speed if I moved to higher overclock settings. 

Is this a software issue, or am I about to hit the physical limits of the silicon lottery?

best regards /Jon

rdugan, 2024-02-28:
neither... the values are averages across all chips, and they can each land at different values near the setting.

Jakkal, 2024-02-28:
> neither... the values are averages across all chips, and they can each land at different values near the setting.

So it’s normal, and nothing to be nervous about, right?

best regards / Jon

rdugan, 2024-02-28:
yep... totally normal, unless the gap is very large.

closed 2024-02-28 by rdugan