rdugan/iceriver-oc #36: Firmware not updated in asic

omining, 2024-03-08:
I have download ks0pro firmware but when I upload the miner resets, it restarts with the default factory firmware and not your firmware

Telicor, 2024-03-10:
It has to be on the latest firmware to update or it will not apply the modified firmware. I had a simular thing happen on one of my KS0s. I removed the modified software because one of my KS0s wouldn't recieve any hashrate, but my other two were fine. Most likely because I have a first batch KS0.

omining, 2024-03-10:
Okay i will try that. So i managed to have it work in one of my ks0pro but it keeps saying machine is offline. Trying refreshing it and disconnecting power but it still says machine is idle even though all lights are green. Do you know why? 
thank you

closed 2024-08-26 by rdugan