rdugan/iceriver-oc #41: Immersion Mode

ColbyAttack, 2024-03-21:
We have a few hundred KS3Ms and have some extra immersion slots right now. We are planning to dunk about 60 of them and was wondering if you any plans for an immersion mode. We have had issues with spoofers in general and would like to just use a dedicated firmware.

Meier3283, 2024-04-07:
I'd second this, just from the perspective of not wanting to see any fan data (in my case I have spoofers), but it applies to immersion as well.  Can we have a way to shut off fans completely, but then even remove the widget on the home page to something else.   Thank you!!

chad5150, 2024-04-07:
Have it your way it at BK😂

Meier3283, 2024-04-07:
You rule! - David Meier

rdugan, 2024-04-07:
> We have a few hundred KS3Ms and have some extra immersion slots right now. We are planning to dunk about 60 of them and was wondering if you any plans for an immersion mode. We have had issues with spoofers in general and would like to just use a dedicated firmware.

As I understand, the issue w/ immersion is the PSU fans, not the asic fans which can be spoofed?  I'm not sure if there's a way to disable PSU fans, but if i get some time i can look into it.

ColbyAttack, 2024-04-11:
Generally, the PSU fans haven't been an issue running in immersion. It's slightly wasteful on power but we have some of the stock Bitmain x19 series PSU's running for several years in immersion without issue. 

I am excited to see what happens to the KS3M's in immersion - hopefully even out the chip temps a little bit.