rdugan/iceriver-oc #43: KS3M

Equadro, 2024-03-25:
hi, on your firmware for ks3m one board stops loading. Only the remaining two work. When I return to the factory firmware, everything works.

myths420, 2024-03-25:
had this same problem spend 3 days trying to get board one working again. resetting and updating to latest ir fw didnt work. after resetting it a few times and updating it finally showed up again.

Equadro, 2024-03-26:
> had this same problem spend 3 days trying to get board one working again. resetting and updating to latest ir fw didnt work. after resetting it a few times and updating it finally showed up again.

What firmware did you restore it on?

rdugan, 2024-03-28:
If you're not changing clocks, there's nothing in my firmware that would affect boards being recognized.  The fact that they work on a different firmware is simply luck - once the hardware goes bad, it can come and go randomly.

You may want to check the cable connections.

KIWIDUDE564, 2024-06-07:
I found that the newest Iceriver firmware will allow you to run the KS3M with bad ASIC chips on a board essentially allowing the miner to cook itself alive. If you download the tswift iceriver monitor you can check your voltages and temps by the chip. If the voltage on any single chip is above .5 you have a bad ASIC chip and it will continue to kill other chips and they are in series so once one chip goes it will up the voltage on the next and so on and so forth till the board is dead.

rdugan, 2024-06-07:
@KIWIDUDE564 my 'alt' ks3m/l firmware is built on the newer stock firmware. It does not 'cook' the machines - it attempts to reduce clocks or stop bad chips from hashing altogether.  

Also, the voltages are already shown in my fw so there's no need for a different tool, and the 500mv story was made up by someone and has never been confirmed by IceRiver afaik.  The voltages are not adjustable on these models, so a higher reading almost certainly just means one or more hash chip is just not functioning properly or drawing the expected amount of power and thus the voltage drop isn't as large as what's seen on a properly running chip.

myths420, 2024-06-08:
im not sure if related. but been doing alot of testing with diffrent pools, ive always used kaspa pool and never had any problems until what happened above and sence then it happened once every few days. i tried k1 pool for about 2 months and board never crashed. went back to kaspa pool and crashed in 3 days then again in 5 days. maybe something weird with that pool?

closed 2024-08-26 by rdugan