rdugan/iceriver-oc #44: KS0 Pro stops mining

Cryptojarus, 2024-04-01:
Hi, I've been using the OC firmware for a few months now. Just got a second KS0 Pro and installed the OC firmware. The miner starts mining but after a while just stops. I used the IceRiver Monitor to see if the chips register an issue, but it all looks normal. It just doesn't mien anymore and keeps running. When I restart the miner, it will mine for a few minutes and stop again. Reinstalled the original firmware and it works for hours on hand. Anyone else have the same issue? I also tried different pools and coins.

rdugan, 2024-04-05:
Are you OC'ing?  If so, try using something less aggressive

Cryptojarus, 2024-04-09:
Thank you for your reply. Yes, I was overclocking but only with 130 and 12. When I use the OC software with 0 and 0 it mines fine. Guess this KS0 Pro doesn't want to run faster. Added extra cooling and everything, all others run well.

rdugan, 2024-04-10:
Every machine is different.  You likely just need more voltage on this one.

closed 2024-04-10 by rdugan