rdugan/iceriver-oc #45: KS0_Pro - Uptime & ChipClock

mrijkhals, 2024-04-05:
Hi guys...... first of all Dugan, thank you so much for this awesome piece code, i really like it!!!!!

But i have 2 questions for you because i can't figure out why things happening.

Q1: The uptime starts randomly over. If it is up for 5min or 20 min and suddenly it is back to 1min. and starts over.
Q2: The chip clock graphic shows a neat green line but with white downspikes. And i just can't figure out why. 

KS0Pro
Clock offset: 137
Voltage offset: 8
Fan mode: fixed speed 64% with 2 out blowing 80mm on the UTP side

Thank you very much in advance. And i am looking forward to future projects

Marc

rdugan, 2024-04-05:
I assume you're talking about the pool uptime.  This means you're having some kind of network issues.  Either the pool is disconnecting, the fee pool is disconnecting, or you're simply having some other network disruptions.  If it's happening frequently, it's like the latter.

For the graphs, make sure you're paying attention to the scale.  A couple mhz movement isn't something to worry about.

mrijkhals, 2024-04-06:
> I assume you're talking about the pool uptime. This means you're having some kind of network issues. Either the pool is disconnecting, the fee pool is disconnecting, or you're simply having some other network disruptions. If it's happening frequently, it's like the latter.
> 
> For the graphs, make sure you're paying attention to the scale. A couple mhz movement isn't something to worry about. If it's dropping to zero, it's just a glitch in the polling - it can happen occasionally if the asic doesn't provide data when requested.

Goodmorning from holland......

About the pool, its this one: kheavyhash.unmineable.com. I reduced the voltage and it's about a little bit better, dont know why.
The graphs is only 1Mhz. 688 to 687 so nothing to worry about... ok:-)

Thanx
Marc

rdugan, 2024-04-06:
It's certainly possible to cause instability w/ too much OC, but from other peoples reports, that usually means the entire machine reboots, not just the pool connection.  I would guess the two are unrelated, but it's not impossible.

mrijkhals, 2024-04-07:
> It's certainly possible to cause instability w/ too much OC, but from other peoples reports, that usually means the entire machine reboots, not just the pool connection. I would guess the two are unrelated, but it's not impossible.

Hi here i am again....
I am nu for 24hours on: eu1.kaspa-pool.org:4444 pool and that goes great. No weird disconnections anymore so that is clear now, it was the unmineable pool.

These are the downward peaks i was talking about. If i keep the voltage low 4, no down peaks but the 3hour avg Hash is 242 instead of 250 (if i follow the calculations). 
If i put the mV up with 1, to 5mV there are some downpeaks like the attached image but the hash goes up.
[image]

Overall i think it looks pretty good now.
clock offset: 138 - voltage offset: 6 - fixed speed fans at 64% - 0 rejects - Diff: 4398.05 G - accepted: 666 ( lol )

[image]

If you think there are some things not in place or weird looking, just ask.

Regards
Marc

closed 2024-04-07 by rdugan

mrijkhals, 2024-04-08:
> those are 1mhz oscillations. The clock is most likely at 687.50, and a tiny .01mhz movement causes a rounding change. It's absolutely nothing to be concerned about.

Thank you very much and keep up the awesome work.....