rdugan/iceriver-oc #46: Machine is offline

zmcochran, 2024-04-05:
after updating the FW and rebooting the miner does not show up in the user interface, it says its offline. I can see the miner running in the iceriver monitor tool though.

rdugan, 2024-04-05:
make sure you have your network set up properly, as described in the 'Known Issues' section of the readme.  if that doesn't help, try a hardware reset (press button for 30s) and then reload the firmware.

ZycoMonk, 2024-07-17:
i have also an issue with "machine down" now on my KS0 pro.. i think everything worked fine, till i decided to move the miner to a different land connected it therefore with via dlan.
if i try to connect to my miner now, the site opens, but after logging in it says "the machine is offline".
if i check the pool the miner seems to be running, as i can see that the miner is doing his 200 GH/s
oh forgotten to say - the miner is running the stock firmware - i can't upload the PBFarmer Firmware in this situation

any idea why this happened? do i have to reset everything?
will i be able to set it all back up then? because if i cannot connect after the reset, the miner won't mine for me anymore

ZycoMonk, 2024-07-17:
oh damn... i ressetet my miner, to update it with the latest PBFarmer firmware, and now it still isn't availible...
still shows after log in "machine is offline" :,-(

ZycoMonk, 2024-07-17:
hmmm... changed the lan cable - still "machine is offline"
did again a reset --> miner was online
so i installed the oc firmware... hope it will stay online now, so that i can make changes to the miner :-/

closed 2024-08-26 by rdugan