rdugan/iceriver-oc #48: Odd clock behavior and a question

Meier3283, 2024-04-11:
I noticed something.  I swore I wasn't going to OC, and just leverage your awesome graphs.   Alas, I couldn't resist, and started bumping it up.  Watching wattage and temps of course.   So, I first did +25mhz.    And the clocks went to 1175 as expected.    I then did +30, and they jumped to 1200mhz.  why would that be?    Today, I went to +35, and they went DOWN, to 1187mhz.  (not 1185).   So it's so far, not linear for me at all.   

Second part is, I know the KS3M's don't have much head room, but what is the range?  What is an average, somewhat safe-ish OC?  +50?  +500?

thank you for any help you can offer.

rdugan, 2024-04-15:
First, clocks aren't necessarily exact - i believe the hardware may work at 6.25mhz increments - and the feedback is from the chip itself, so that and the setting may slightly mismatch. As for the +30 going to 1200mhz, my guess is it's because there are two competing clock setting mechanisms - the direct one i wrote, and one from IceRiver that works in larger increments with it's own slower ramping cycle.  I've tried to make sure the IceRiver one doesn't override mine, but it sounds like you got caught between them somehow - maybe you adjusted the offset while it was still ramping or something.

I have no idea on range on any of these.  I don't own any, and there are no published specs on the hash chips.  Any information i get is from public forums where other people talk about their tests.  Additionally, I've seen massive variations in reports of stock power draw, so I really don't recommend OC unless you have a power meter, in which case, you simply need to watch that to know your headroom.   Assuming the 3400w rating is correct, and these are platinum rated, and reasonable quality (risky assumptions on both accounts w/ IRs track record), then I would guess you're around 90% efficiency at max load, or ~3750w at the wall.

Meier3283, 2024-04-15:
Thank you!  In my case, I can report that when I run at 1200mhz, it's pulling right at 3300watts.  Works out to around 6500/th.  for whatever it's worth.

I did just have my miner stop reporting intake temps, on one of the three boards. Weird.  Rebooting and monitoring.
I appreciate your help!

rdugan, 2024-04-16:
No problem.  3300w sounds quite low for being overclocked, unless you're using external fans?  If not, I wonder if your issues are a sign it's simply not getting enough power for those clocks...

Meier3283, 2024-04-17:
Yes, I’m on AC infinity 8” fan.   - David Meier

rdugan, 2024-04-17:
Ok, that makes more sense then.  So if you're still seeing the reporting issues after the reboot, maybe try a lower clock offset first.

Meier3283, 2024-04-17:
It rebooted an hour ago, so I just set the clocks back to stock for now.

closed 2024-08-26 by rdugan