rdugan/iceriver-oc #49: Password Error after reseting miner

nordman666, 2024-04-14:
is it only me?

have installed your firmware some days ago and today i got password error after i hard reset the miner

my changed PW wont work and also the standard 12345678 wont work.

all LEDs on the miner are allways on now, reset agaion and miner reboots and still all LED on

Meier3283, 2024-04-15:
I had this happen yesterday.  I KNOW what my password was.  did a reset to get around it, and even then, neither the password I set, nor the default password would work.   I hard reset again, and was able to get into the default, and get it configured.   It was weird.  Up and running fine now.   I'll complain if it happens again.   (Separately, I did have a reboot that day, and I don' know why.  Not sure if there are logs I can review, or not)

rdugan, 2024-04-15:
There are certain files that IceRivers shitty software seems to occasionally corrupt or completely erase, for unknown reasons.  One of them is the config file that they store your password in.  I've rewritten their entire auth routine to both fix this and get rid of the gaping security holes they've created, but until I release it, the only option is to try to hard reset numerous times until it works again, or go into the file system (via tcf or ssh) and manually replace the files.

nordman666, 2024-04-16:
> There are certain files that IceRivers shitty software seems to occasionally corrupt or completely erase for unknown reasons. One of them is the config file that they store your password in. This happens regardless of whether you're using my firmware, but it could be possible that crashes from bad OC, or more frequent resetting may trigger it.
> 
> I've rewritten their entire auth routine to both fix this and get rid of the gaping security holes they've created, but until I release it, the only option is to try to hard reset numerous times until it works again, or go into the file system (via tcf or ssh) and manually replace the files.

iceriver batch tool can reset the device at first try :)

but...did u have the SSH password?! would be great to get it to have ability to backup and restore config this way

rdugan, 2024-04-16:
Yep, sorry... should have mentioned the batch tool.  And no, i don't have the ssh password.  For KS0Pro, i provided a firmware update that adds a new account w/ ssh access.  For the other models, you'll have to do it yourself manually via the open TCF port.

nordman666, 2024-04-16:
which files i need to back and restore cause again every 24hours my miners lost temp sensors and hashrate went down to 2-3th and afterwards i got the error with non working password. is there also a way to check why every 24hours the miners have this problems?
i use awesome miner - a mining monitoring and management software and i can send reboot commands via webinterface and this works with set password - so it doesnt seems that the password is gone completely - just the login page dont accept the password anymore

i hope i can provide some info that you maybe able to fix that issues

rdugan, 2024-04-17:
It's the config file at /var/order/conf/bgserver.conf iirc.  And the password is gone - the API doesn't need credentials (one of the many security holes of these things).

But yeah, sounds like you have bigger problems, and this is just a symptom.  Bad power supply maybe.  These file issues remind me of FS corruption that used to be common on PCs after power loss or improper shutdown.

Meier3283, 2024-04-17:
Looking forward to your next release as well :)

nordman666, 2024-04-17:
> It's the config file at /var/order/conf/bgserver.conf iirc. And the password is gone - the API doesn't need credentials (one of the many security holes of these things).
> 
> But yeah, sounds like you have bigger problems, and this is just a symptom. Bad power supply maybe. These file issues remind me of FS corruption that used to be common on PCs after power loss or improper shutdown.
> 
> Btw, like i said, i've already rewritten the auth, so the password problem is fixed - just haven't released it yet.

no..you dont understand. it is exactly after 24hours the miners (i have more than one) will hash only 50%, sensors shows weired temperatures and password is gone. i need to reboot with icedriver tool and than it is running again on pool and with full hashrate (i set -100mhz ofset).

rdugan, 2024-04-17:
Not sure what you're saying i don't understand.  I'm saying the missing pw is not causing your problems - your problems are causing the missing pw.  You have some kind of hardware/power/network issues, and the corrupt file is just a result.

Meier3283, 2024-04-18:
So, I’m getting the error again after 23 1/2 hours. I went to login on my desktop and got the password error.  Took these screenshots from my phone.  Logged out of my phone session, and got the error trying to log back in. 

[image]

Meier3283, 2024-04-18:
[image]

rdugan, 2024-04-18:
Based on the timing, my guess would be you have some sort of regular network interruption.  

There's nothing i can do about these password errors... I don't touch that config file, it's all handled by the IceRiver miner software, so either it's their bug, or the result of external factors (or both).

The only option I have is to replace the auth mechanism entirely - which as I said, I've already done.  If you want access to the prerelease, find me in the Kaspa discord.

Meier3283, 2024-04-18:
Ok, sounds good. I joined up and will post when I can.

richm4n212, 2024-04-19:
> is it only me?
> 
> have installed your firmware some days ago and today i got password error after i hard reset the miner
> 
> my changed PW wont work and also the standard 12345678 wont work.
> 
> all LEDs on the miner are allways on now, reset agaion and miner reboots and still all LED on

Im also like that i think they stole our hasrate because power stiil same n they change our pw so we can check it

richm4n212, 2024-04-19:
> I had this happen yesterday. I KNOW what my password was. did a reset to get around it, and even then, neither the password I set, nor the default password would work. I hard reset again, and was able to get into the default, and get it configured. It was weird. Up and running fine now. I'll complain if it happens again. (Separately, I did have a reboot that day, and I don' know why. Not sure if there are logs I can review, or not) to get my to reset fully, I think I held the reset button for like 30 seconds.

They steal your hasrate n change the pw i see the pool n wallet change after pw canot acces 🤣

Meier3283, 2024-04-19:
Happened to me again, I didn't catch it right at 24 hours, but It's been up 25 hours.  Temps went crazy. Still hashing, and hashing to the proper address. I don't think anything is being stolen.  No way they have access to my miner.    I've been pinging the IP all day, and there aren't any gaps, at least not more than an occasional single lost packet.  I also went back to factory firmware yesterday, so certainly not firmware specific.   I'm just reporting, not asking for a fix.

rdugan, 2024-04-19:
> > is it only me?
> > have installed your firmware some days ago and today i got password error after i hard reset the miner
> > my changed PW wont work and also the standard 12345678 wont work.
> > all LEDs on the miner are allways on now, reset agaion and miner reboots and still all LED on
> 
> Im also like that i think they stole our hasrate because power stiil same n they change our pw so we cant check it

It's not unheard of.  There have been reports in the past few months that people were hacking into machines and redirecting hashrate.  These things are incredibly insecure.  But if you're not forwarding any of the ports or using something like hiveos, then it's not really possible.

acs999, 2024-04-21:
@rdugan 
I can confirm these Problem with the original software. I had to hard reset it and reconfigure it already twice within 3 month.

rdugan, 2024-08-01:
This should no longer be a problem in my latest firmware, as i've replaced IceRivers broken authentication processes.

closed 2024-08-01 by rdugan

Nolan20007, 2024-08-30:
tell me have you solved the problem somehow? I have the same error running for 20-24 hours, the temperature of the board jumps from 65 to 127 degrees instantly, and at this moment the hashrate is about the 2nd-3rd, and does not allow Asika to enter the muzzle (not the correct password), iceriver ks5l