rdugan/iceriver-oc #50: 12 hour average?

SomethingGettingWrong, 2024-04-18:
2 hour is not enough. Can you please do 12 hours? id ratehr replace teh 5 min. with 30/2hour/12hour if you cant hav eall 4

rdugan, 2024-04-18:
2hrs is absolutely enough.  It results in a ~0.5% variance.  Time is not what matters, share count is.  And the share count in 2 hours in the UI is still an order of magnitude larger than anything you're getting from a pools 24hr reading.

Also, there is limited memory on the asics, and storing 24hrs of data would be problematic if not impossible.

closed 2024-04-18 by rdugan

SomethingGettingWrong, 2024-04-19:
I am defintily not smarter then you. You know your stuff. its self-evident. However, i would like to point out for future reference. that that share count at the same clocks... wiht diffrent MV at 2 hours... ive had to restart miner 4 x in order to get a better picture.

For example  2 hours at 32 mv 325 mv showed  same hashrate as  2 hours at 34mv at 325 mhz  

However when i reset the thing every 2 hours 3 x linerally for 32 mv at 325 mhz  and  34 mv at 325 mhz i noted 7 more gh/s that didnt show up at 2 hours at 2 hours it was the same multiple times.  it was only after i went about 6 hours did i see the diffrence. I just threw 12 out there. 

Again its neglitble so doesnt really matter but I did notice.  taht the shares averaged out higher on 34 mv at 325 mhz then 32 mv at 325 mhz over a longer perido of time where as 2 hours it was the same. 

maybe my lack of understanding. Thanks bro.

rdugan, 2024-04-20:
I'm not quite understanding.  Are you saying you saw 7ghs more on the pool after 6 hours?  If so, most pools generate probably somewhere around 1800-2160 shares in 6hrs, giving an expected variance around +/- 8-10%.  As a result, a 7ghs change around 318ghs or 2.2% delta is statistically indistinguishable from noise.

The ASIC runs it's own internal diff independent from the pool diff for the hashrate measurements, and 2hrs in the UI is two orders of magnitude more shares than the pool - likely somewhere around 500,000, which results in an expected variance of roughly +/-0.5%.  So that same 2.2% delta in hashrate can be easily distinguished from 0.5% noise after 2hrs.  

Adding a longer average would not help to identify a 7ghs/2.2% variation in any way - in fact it would do the opposite.  The only thing it would help is to identify some variation less than 0.5%, which to me seems unnecessary, as simple things like network disruptions or even ambient temperature changes can easily swing your hashrate more than 0.5%