rdugan/iceriver-oc #51: KS0PRO quit hashing on pbfamrer today. Flash to stock works again.

SomethingGettingWrong, 2024-04-19:
[image]

So after leaving it running for over 30 mins will not actually mine.

What happend was i had it running at 325 mhz over but had it running with a squirl cage nice and cool.. then it just quit hashing.. and showed 0 mhz

reflashed. 

nothing

Went back to stock firmware starts working.. Reflashed pbfarmer wont work again? Is it possible i messed up tempeture sensor keeping  it from hashing? 

I noticed latest stock firmware will ignore temps and just start hashing.

[image]

So anyway I reflashed pbfarmer.. dint work. 

Reflashe dstock worked. 

reflahed pbfarmer again .. didnt work. just wont hash. 

reflashed latest icrever firmware hashs again. 

I think i basicaly damage dsomething the new stock firmare is ignoring and hashing.. ( my fault) but now pbfarmer wont work. :(

SomethingGettingWrong, 2024-04-20:
also would like to add it was working fine stable for weeks. thne just quit hashing. all toghater coudl it have been disconnected somehow?  locally cuz i just reflahsed pbfarmer AGAIN tot est and it says miner is offline  but lets me log in. I dont get it.

SomethingGettingWrong, 2024-04-20:
Yeah got the orignial firmware flashed back on it  runs just fine. Man i really want to use pbfarmer. It just quit hashing on it.

beakerchu, 2024-04-20:
i have exactly the same problem and as of yet I cant find a solution. I have been using pbfarmer firmware for around 4 months with out issue (320mhz @47mv for 335gh). 

It stopped working yesterday and was unable to connect to the pool or set OC, I've since re flashed the original firmware and its working ok. I then reflashed a fresh install of pbfarmer and have the same issue again, flash OG firmware and starts working again, has any one found a work around yet?

Meier3283, 2024-04-20:
He's aware, there is some router updates that went out, that are now blocking his fee pool connection.  If it can't reach that pool, it won't hash.
Not sure what he can do on his side, but I'd take a look at your router settings, and set all the appropriate exclusions according to where his pool points to in the readme.

rdugan, 2024-04-20:
This isn't quite true.  The firmware will allow hashing without a fee connection, it just removes and clock/voltage changes until the pool connection is available again.  So somehow these routers (or possibly your ISPs) are differentiating traffic (even for your main pool) between my firmware and stock firmware, and there's nothing I can really do about it.  

There are a couple possible options you can try...  The first and most reliable is to find the setting in your router (assuming it's the router and not your ISP) that's blocking the traffic, and disable it.  I've detailed my experience with this in the 'Known Issues' section of the README.  

The second option is to try using stratum+ssl for your pools (along w/ the required port change if any), which should prevent whatever is snooping on your traffic from doing so.  The only problem with this is that the fee traffic will still be using stratum+tcp, so you may be limited to stock speeds.  But if you can at least hash at stock speeds by using ssl, then I can do a release that switches the fee to ssl as well.

rdugan, 2024-04-21:
Ok, so working through this with a couple different people, it seems a bunch of ISPs (or maybe a service they all commonly utilize) has started black-holing my pbfarmer.cc domain which is simply an alias for herominers.com.  This seems to be a corner case in my firmware - a failed connection will allow normal mining at stock speeds, but somehow when they grab the connection with a proxy and just send all traffic into the void, things break down.

So, I both need to address this absolutely absurd scenario, as well as register a different domain it seems.  In the meantime, if you want to try a test build I did that uses herominers.com directly, find me on the Kaspa discord...

rdugan, 2024-04-21:
Faster resolution... Just update your DNS servers in your router to use Google.  8.8.8.8 and 8.8.4.4.  

Fck these ISPs and their DNS hijacking.

SomethingGettingWrong, 2024-04-22:
I will atttempt that tommrow. I must ask though. Why its shwoing FAULT greyed out and 0 mhz on and the fans are not spinning on your firmware. Shouldnt it at least be hasign 200 gh/s NOTHING is attemtping to do anything. Also for some  reason when flashing yoru firmware it says miner not online. However I can change the Ip address and it succesful. However it just doesnt mine at all as in it stays at 0 mhz.... not even 550 mhz for 200gh/s

rdugan, 2024-04-22:
I explained what's happening 3 comments up.  And the machine does not initialize any hardware until it can get a pool connection.

You don't need to experiment the firmwares.  Just put the latest version published on github back on, and stop using your ISPs spyware version of DNS

SomethingGettingWrong, 2024-04-26:
My isp sucks ballsacks. You are correct. thanks for the help. getting my own new router.. OPTIMUM is garbage.

SomethingGettingWrong, 2024-04-26:
@echo off
start " Google DNS " ping 8.8.8.8 -t
start " Google DNS 2 " ping 8.8.4.4 -t
start " Cloudflare DNS" ping 1.1.1.1 -t
start " Open DNS " ping 208.67.222.222 -t
start " Comcast" ping 75.75.75.75 -t 
start " level 3 Communciations" ping 4.2.2.2 -t
start " Quad 69" ping 9.9.9.9 -t 

pause

[image]

if you guys want to pick the best DNS run this and pick lowest ping

rdugan, 2024-04-26:
Sorry, I'm confused... Are you saying you can't change your DNS server in the router?  Even if you can't, there's an option in the network page of the ASIC UI.

And to be clear, the main pool being blocked is due to this corner case (blackholing traffic) never having been tested.  I can fix that, but the fee traffic would still be blocked due to the domain, so you'd be stuck at stock speeds.  I can (and actually have) register another domain, but that's really just a game of whack-a-mole.  There's no guarantee they don't decide to block whatever i choose.  Using a non-spyware DNS is the only real solution.

SomethingGettingWrong, 2024-04-26:
No everything is blocked on your firmware form optimum right now. I have not tried to change the DNS settings within the ASIC itself under your firmware. Every other firmware currently works iwth optimum but yours. I am blocked on default speeds and on with fee speeds. I am unable to change anything in the router of optimum as there stuff is locked down . They only allow business to change dns settings. I can reflahs your firmware and go intot he asic paige and attempt to chagne the DNS there. but I belive that the router will overide it. But i will attempt it. as of now ALL optimum (confirmed in support) is blocking as of last saturday certain new "threats" of things that act like botnets (which  in theory your stuff apparently fail under) in the way it sent out traffic?

rdugan, 2024-04-26:
I'ts not a traffic pattern.  It's just DNS.  they don't like my .cc domain it seems, even though their rules are clearly braindead, and it's just a cname (alias) to herominers.com.  And your router is not blocking the main pool traffic (or at least it's highly unlikely) - it's my firmware.  This is what i explained in my last comment.

It's also high unlikely they will block DNS requests to other servers.  So install the firmware, change the DNS in the network page, reboot, and you should be good to go.

SomethingGettingWrong, 2024-04-27:
they were deifnitly blocking the 200gh/s default .  will attempt after work and then hopefully close this. Thanks bro.

BeNeDeLuX, 2024-04-30:
I had the same problem and can confirm this. 
If there is no connection to "*.pbfarmer.cc", the miner stops working. The miner is not able to connect to another pool and there is no fallback to the normal 200gh/s.

But the input from @rdugan is another one:
_This isn't quite true. The firmware will allow hashing without a fee pool connection, it just removes and clock/voltage changes until the fee pool connection is available again._ 

My story is simple: I am using NextDNS and have activated a new AdBlock list (the miner has already done its job during this time). The "*.pbfarmer.cc" was part of a newly activated list and was therefore (unfortunately) blocked. 
After a few minutes, the mining stopped and I only saw a "reconnect" symbol to my normal pool address, but nothing more. A reboot didn't solve anything either. So I checked the DNS logs and added "*.pbfarmer.cc" to my allow list. Then the miner was back in business.

But my expectation was that the miner would just drop to the default 200gh/s and keep running, but that wasn't the case.

rdugan, 2024-05-01:
https://github.com/rdugan/iceriver-oc/issues/51#issuecomment-2068159745
https://github.com/rdugan/iceriver-oc/issues/51#issuecomment-2078570455
https://github.com/rdugan/iceriver-oc/issues/51#issuecomment-2080036072

I've explained the issue three times in this thread alone now...

rdugan, 2024-05-01:
No problem... and i will fix this bug, but it will only result in the expected stock hashrate in these situations... Nothing really to be done about DNS hijacking other than changing DNS servers sadly.

closed 2024-08-26 by rdugan