rdugan/iceriver-oc #55: ks0 pro: no connection to the poolwith oc firmware  / with stock firmware connection works fine

ZycoMonk, 2024-05-13:
hello together,

my ks0 pro works fine on stock, but as soon as i switch to the pbfarmer oc firmware (thanQ @ rdugan) the ks0 pro doesn't connect to the mining pool.
i used the version v0.71 

i tried so far:
- switching off all kind of router filters and firewalls 
- my ks0 pro can open all ports needed by itself
- else i am using cloudflare-dns 1.1.1.1 

does anybody know how to solve this problem?

[image]

rdugan, 2024-05-13:
try changing your DNS server (preferably in your router) to something other than the ISP provided one.  For instance, Google's servers (8.8.8.8 and/or 8.8.4.4)

ZycoMonk, 2024-05-13:
changed the dns to google's 8.8.8.8 / 8.8.4.4.
no change it still doesn`t connect to the pool...

but what i noticed, the first standard connection that comes with the first oc firmware installation works like a charm..
[image]

change it back to my pools settings end in...
[image]

ZycoMonk, 2024-05-13:
just to proof.... switched back to stock firmware, and it directly connects to the pool... ¯\_(ツ)_/¯
[image]

rdugan, 2024-05-14:
Yes, this is why you're changing your DNS.  It's almost certainly the same issue everyone reported 2 weeks ago.  A number of ISPs started blocking the .cc TLD, which is what my domain for the fee traffic uses.  

See https://github.com/rdugan/iceriver-oc/issues/51

Are you sure your asic is using the correct DNS server now?  Did you reset it to dhcp in the network settings and reboot after you changed the DNS servers?

rdugan, 2024-05-14:
Ah, sorry.  I misunderstood what you were saying in [this comment](https://github.com/rdugan/iceriver-oc/issues/55#issuecomment-2109013180), and the image links were dead at the time, so i couldn't enlarge them.  

This is unrelated to my firmware.  Kaspa-pool is down.

closed 2024-05-14 by rdugan

ZycoMonk, 2024-05-14:
wait

ZycoMonk, 2024-05-14:
https://github.com/rdugan/iceriver-oc/issues/55#issuecomment-2109022580

restarted last night, but wasn't testing, because my router too so long... ^_^
today kaspa pool is connecting with the oc firmware... so may be it was because of cloudflare dns 

thanQ R. Dugan (or Erdogan?)