rdugan/iceriver-oc #56: Job rate and uptime pool and accepted

Jerobech, 2024-05-15:
I have noticed that my job rate is 0.47  is this oke? 

And a other thing i have noticed that the uptime from the pool is reset back to zero once in a while also my accepted gets back to zero.

Do you know why?
Is my latest point caused by mining to your account once in a while?

Best regards
Jero

rdugan, 2024-05-15:
Job rate should be ~1/s currently, as Kaspa block rate is ~1/s.  Unless you know the pool is purposely rate-limiting jobs (e.g. kaspa-pool), then 0.47 is not 'right'.  But it doesn't necessarily affect your earnings either, as kaspa doesn't have 'late' blocks like normal chains.

Uptime and accepted are reset because your connection was interrupted.

closed 2024-05-15 by rdugan

Jerobech, 2024-05-15:
I am using humpool.com

Job rate is comtiniuos about 0.47

rdugan, 2024-05-15:
You should ask them why they're not sending all jobs then.

rdugan, 2024-05-16:
I'm not really sure.  I don't do anything w/ hostname.  You can always set up ssh access and change it.

rdugan, 2024-06-07:
@Jerobech First I will need to work with someone that has one, so I can't do anything before they ship.  Also, iceriver changes things with every new model, so unfortunately it's not as simple as you might think.  I'm still trying to work through their changes for the ks5l.

Jerobech, 2024-06-07:
Thanks for your reply,

Please can you give me an update when you are starting on the ks0 ultra?

Best regards

Jeroen

rdugan, 2024-06-11:
I can't do anything before they ship.  Also, based on what I'm seeing on the KS5L, it doesn't look like it will be possible to replace the firmware with anything not from IceRiver themselves.

Jerobech, 2024-06-11:
Here in holland some guys already have the ultra at home