rdugan/iceriver-oc #58: ks0 Ultra

pacmenDE, 2024-06-26:
@rdugan What's abou the new ks0 ultra? Is there a plan to overclock the machine :-)

JoeyTheKasMaster, 2024-06-29:
Yes I second this!!  We need the kso ultra firmware PB!!!!! =)

craigc403, 2024-06-29:
third this.... we need this!!!!!

rdugan, 2024-07-02:
I don't have one yet (I'm in July 1-15 batch), but I would assume, like the KS5L, that IR has started cryptographically signing firmwares, meaning you can't really install any third party ones, without jumping through major technical hoops.  If something changes, it will be released here as usual.

JoeyTheKasMaster, 2024-07-02:
Dangggg that's sad 😔   weren't they on board You and t swifts firmwares
the first time?

Jerobech, 2024-07-02:
major technical hoops.  Is this the level normally your skills stops?

qidao2, 2024-07-03:
I hope KS0 Ultra firmware can be released earlier

rdugan, 2024-07-03:
> Dangggg that's sad 😔 weren't they on board You and t swifts firmwares the first time?

They weren't sanctioned by IceRiver, if that's what you mean.

rdugan, 2024-07-03:
> major technical hoops. Is this the level normally your skills stops?

It's not a question of my skills.  It's a question of the ability of 99% of users not being able to actually get it on their machines, when it requires extra hardware, disassembling the ASIC, and familiarity w/ linux

Jerobech, 2024-07-03:
I have talked to tswift today, he claims he can do it.  Cryptology is not
his main concern he said.

Mvg
Jeroen

chad5150, 2024-07-04:
@Jerobech 
Not sure this is the platform for a comparison of skills. If you’re not offering solutions or intelligent conversation please troll elsewhere.

Jerobech, 2024-07-04:
> > major technical hoops. Is this the level normally your skills stops?
> 
> It's not a question of my skills. It's a question of the ability of 99% of users not being able to actually get it on their machines, when it requires extra hardware, disassembling the ASIC, and familiarity w/ linux

I am not trying to offend you.  Sorry if it sounds that way. 
Maybe it is option and a business case for you that customers send you the unit, you customise it and send it back

bm3-maker, 2024-07-13:
Any News for the Ultra Firmware? How its going?

MathijsNL, 2024-07-15:
> I don't have one yet (I'm in July 1-15 batch),

@rdugan I have a few running at the moment. Anything I can do to contribute?

craigc403, 2024-07-18:
looks like Overclock firmware is out by tswift for the ultra. however i prefer pbfarmer where i can control voltage/mhz..

bm3-maker, 2024-07-18:
Me too. I wait for Pbfarmer. I think it won't be long now

bm3-maker, 2024-07-31:
is the new 0.8.0. working at Ultra?

Fillelind, 2024-07-31:
pbfarmer released one now, tried it out on mine(Fan shroud 12v standard case fan + same fan on side, 100% internal fan, no powerstage mod)

Clock offset: 200Mhz
Voltage offset: 153 Mv
Power draw is at 200W

Hashrate is about 618-650 Gh
Powerstage temp tops at 115⁰C
Chip temp tops at 72⁰C

craigc403, 2024-07-31:
Just did mine, seem to brick it... No web interface, however i can it hashing on defaults through monitor.. Ive tried everything, but web interface will not come up. any ideas?

chad5150, 2024-07-31:
Same issue here. After firmware update machine will not hash. After many factory resets and updates I have it hashing at 100 gh. The interface is stable just won’t hash. On factory reset I get “machine not online” error so I can’t change to a pool that I can look at and see if it’s hashing

Jerobech, 2024-07-31:
Use https

craigc403, 2024-07-31:
ok mines working now.. factory reset it again using button , hold down for about 20 seconds while machine is up, until both lights blink... it will reset.. let sit for about 10 minutes, finally came up.. 
reuploaded PB 8 firmware, let reboot, be patient, take extra time to come up, took mine 5 minutes.

clock offset 150
voltage offset 100

hashing around 550.
i have copper heatsinks , internal fan mounted on outside pointing in at 100 percent. pushing air inside with 2 slower fans on top and bottom, not the good ones..

Chips temps mid 60s
power stage around 83

Jerobech, 2024-07-31:
Did you use https  in the browser ? i have read something in the readme file?  

And what is your wattage at 550 ghash?

craigc403, 2024-07-31:
no non secure.. also about 160 watts

rdugan, 2024-07-31:
You can't use non-secure.  It's not possible - the firmware (web server) will force https.

craigc403, 2024-07-31:
ya the browser is warning me.... i dont mind it..

[image]

rdugan, 2024-07-31:
Ok.  Just to clarify, it's still using https and is 'secure'.  The warning is simply because the certificate is not issued by a known authority.

rdugan, 2024-08-01:
closing this now that the firmware has been released.

closed 2024-08-01 by rdugan

Hienishere, 2024-08-01:
I need help I have a firmware upgrade but something went wrong and now I can't access its ip page anymore can someone help me

rdugan, 2024-08-01:
Check the documentation:
https://github.com/rdugan/iceriver-oc?tab=readme-ov-file#known-issues

Also, please open a new issue if you still have problems.  This one has been closed

Jerobech, 2025-05-04:
Hello,

I have an question,  do you think we can use the iceriver miners for the
new BLOCKdag  coin or does the firmware also need to be modified

BLOCKdag uses the same algo as Kaspa i think

Best regards

Jeroen