rdugan/iceriver-oc #59: WebApi for Power Management

pacmenDE, 2024-06-28:
Hello @rdugan,
I would like to program a small app that puts the miner in sleep mode and switches to work mode when needed.

Do the Icervicer miners have such a WebApi connection?

Such an app would be particularly interesting when using a solar system or night-time electricity tariffs... :-)

nordmann666, 2024-06-30:
they dont

just use a PDU with API support and you can do what u need to do

rdugan, 2024-07-02:
I have a pre-release version w/ a reworked API among other things, that includes sleep/wake endpoints.  However it's in a build that is intended for commercial installations (includes a few other features most end users don't need), and has a slightly higher fee (1.33%).  It will be ported back to the 1% version when it's deemed stable, but if you want to help beta-test the current version, find me in the Kaspa discord...

SimonFischer04, 2024-08-24:
Hi @rdugan 
The new API looks very promising, thanks :+1:. 

Would it be possible to provide the json (https://github.com/rdugan/iceriver-oc/blob/main/docs/apidoc.json) in a Swagger/OpenAPI format, so it is easier to consume for client implementations?