rdugan/iceriver-oc #62: pre-paid version

BahmaNvidia, 2024-07-10:
hi
do you have a pre-paid (non dev-fee) version of these firmwares too?
i have a non resolvable problem with dev-fee switching between pools that causes invalid shares all the time

rdugan, 2024-07-10:
I don't.  But the firmware doesn't switch for dev fees - the connection is maintained in parallel 100% of the time.  So the problem is likely unrelated, or at least resolvable.  Maybe you can provide more details?

BahmaNvidia, 2024-07-10:
I have set static ip for the miner in the router and with the same internet connection an same pool etc I don't get a single invalid share for days with the official firmware or pre paid overclock firmware from tswift , but with your firmware (especially version 0.7.0 and 0.7.1) the pool connection time resets every few hours (and sometimes in less than an hour) and I get invalid shares constantly. though the 0.6.0 version gives much less connection resets (maybe once or twice a day) but invalid shares are still there. even if I don't increase the clock and voltage.

rdugan, 2024-07-11:
that's because nothing else is telling you about the connection resets.  i added that in later versions.
earlier versions will show invalid shares for the fee, which is not the case in the latest version

closed 2024-08-01 by rdugan