rdugan/iceriver-oc #65: Kas0 Pro Board1 Power Stage

1Knightwing, 2024-08-01:
Where exactly is the subject located on the device (board1 power stage)? I’d like to try and use some heat sinks on it to get the temps down, running too close to 125* for comfort!

rdugan, 2024-08-01:
It's the 4 black chips in front of the chokes (tall boxes).  The ones that everyone (including most youtubers) mistakenly refer to as mosfets (there are two mosfets inside the power stages).

closed 2024-08-01 by rdugan

1Knightwing, 2024-08-01:
Thanks! Got a heat sink (6x6x5mm) on each of those already so guess I’ve done all I can to mitigate that issue, apart from maybe stronger fans. Temps there range from 108-121* C, with clock at 831mHz for a stable 300-320 gh/s, my goal range. I gather they can operate at 125* C.