rdugan/iceriver-oc #67: Ks0 ultra when OC applied stops mining

chad5150, 2024-08-03:
When i apply an oc the device stops mining. Even a 1% increase will tank it. The OC settings have to be at 0 for it to mine. With the firmware it mines around 300 gh. Stock fw mines around 380. I have a 185 watt power brick also.  
Thank you.

closed 2024-08-03 by chad5150

MinerGuy62, 2024-08-06:
I have this issue also and I get ASIC Request failed often with pbfarmer loaded.  Any OC applied it will stop completely.  Is this something that will be fixed in a later version?  I have a KS0 Ultra

rdugan, 2024-08-06:
@MinerGuy62 please create a new issue, or just find me in Discord.  This one was closed by the author - I'm assuming it's been resolved in his case (this is unlikely to be an issue with the firmware, but rather a power/temp issue.)