rdugan/iceriver-oc #69: KS3M

mb-pro69, 2024-08-07:
Hello, i installed your firmware on my ks3m today but it doesn't run, i can't access now to asic configuration page. I use static IP for this asic. If i do a hard reset and come back to the official firmware, and try to access again to configuaration page (same static IP adress), it's ok, no problem !
But with the OC firmware impossible to connect me...
I've read the known issues in README file but it haven't helped me.
Can you help me please ?

mb-pro69, 2024-08-08:
Hello rdugan, first thanks you for your answer and your help. Excuse me, english isn't my first language... I mean, it uses a static IP in my router (FAI) but, in asic settings, it's on DHCP... I've already try mozilla and chrome bronwsers.
 I don't know wat i can verify more?
[image]
[image]

mb-pro69, 2024-08-08:
with official firmware, i access again to configuration page:
[image]

rdugan, 2024-08-08:
There are only two reasons I know of why you wouldn't be able to access the page.  The first is that it's the wrong IP, but your setup looks like it should be ok.  

The second is that my firmware is now using https, and the TLS certificate is generated on the machine, rather than by a known authority, so it will be untrusted until/unless you import the CA certificate into your browser.  But in that case you should still have a warning in your browser that allows you to 'Accept the Risk and Continue' - which is clearly not present your case.  

Try 'force refreshing' the page (i.e. load without cache), or clearing your cache then reloading.  If that doesn't work, I can only suggest to do a hardware factory reset (hold the button for 40s) then reupload the firmware, in case the machine is still using old network settings.

rdugan, 2024-08-08:
Also, I should mention that there is a bug in the KS3M clock setting mechanism, so if you were using this to over/underclock your ASIC, i would recommend using 0.7.1 until I can release an update.

mb-pro69, 2024-08-08:
Good evening and thank you for this information. I just did a hard reset, I emptied the cache, I tried to open the page in a private firefox window, I emptied the cache but nothing to do, it's strange :(
[image]

mb-pro69, 2024-08-08:
the firmware i am trying to use is 5977449_ks3m_miner.bgz

rdugan, 2024-08-08:
I have no idea what that firmware is.  It's not mine.  So it won't be using https

mb-pro69, 2024-08-09:
no it's the firmware i found in your Github, you affraid me, i verify now

mb-pro69, 2024-08-09:
Isn't it this firmware i may use ?
[image]

rdugan, 2024-08-09:
No.  that's "extras" - ie related software, not mine.  As the folder name says, these are the original firmware files from IceRiver.  

My firmware is in the releases section:

https://github.com/rdugan/iceriver-oc/releases

mb-pro69, 2024-08-09:
ok sorry, but in these firmwares, which one is for ks3m ?
[image]

rdugan, 2024-08-09:
There is no 0.8.0 version available at this time.  I've removed it because of the bug I mentioned above.  You can use 0.7.1 until I release the fix in 0.8.1.

mb-pro69, 2024-08-09:
i am so sorry, so i was using the bad firmware :( Thanks a lot for your help. I am going to try it:
[image]

 have a nice day ;)

rdugan, 2024-08-09:
No problem... I'll update/close this issue when i release the new version.  Cheers!

rdugan, 2024-08-26:
New release (v0.8.1) has been pushed, including KS3L/M builds

closed 2024-08-26 by rdugan

mb-pro69, 2024-08-28:
Good job. Thanks fo all :)

rdugan, 2024-08-28:
There is no tuneable power circuitry on KS3* or KS5* models.

mb-pro69, 2024-08-28:
ok, you know better than me :) Thanks