rdugan/iceriver-oc #72: Fan control buggy?

Lyxon1337, 2024-08-15:
KS0 Ultra Standard fan replaced with 2x "NOCTUA NF-F12 5V PWM"
(Tunnel structure for cooling)

Updated KS0-Ultra stock firmware -> pbv080_ks0ultraupdate.bgz

All working fine for days with Fan Configuration "Target Temp"

After Shut down -> KS0 installed at its final location -> Power on = Red LED blinking and KS0 not work.
Even if i setup "Fixed Speed: 100% -> not work (but the RPM are shown!)

Wazuup here? is there any easy fix?

rdugan, 2024-08-16:
considering you replaced a 6000 RPM fan w/ a 1500 RPM fan, and teh miner tests for at least 3000 RPM from what i've seen, i'm not sure how you ever had it working.

Lyxon1337, 2024-08-16:
i do a reset and now it works again... the stock firmware is green on "fan speed".
and yes, on boot it is red... but if i set the fan speed to 100% it turns green and works (after that i can even set 0%, it is still green).

On the pbv080 it's always red :/ I dunno why it's works on first upgrade

rdugan, 2024-08-16:
because you're resetting to stock v14 which doesn't check fan speeds.  stock v15 (which my fw is built on) does

Lyxon1337, 2024-08-16:
and it is not possible to adjust this fanspeed check from the hardcoded 3000 value?

Like... a window where you enter the maximum speed of the corresponding fan.
And from that it draws its own "check" speed.

Or a check mark that confirms that it only gives an error at 0RPM or when none is plugged in... and only then gives the error? (assumes that a minimum RPM has been set)...

or is that idea a complete no-go?

rdugan, 2024-08-16:
no.  this is in the IceRiver compiled miner binary.  I can't add features/UI to that.  Also, 0RPM test isn't really sufficient.  because a broken fan that only does 100 RPM for instance would pass, but be effectively useless.

Lyxon1337, 2024-08-16:
yeah sure i get ur point...
but right now with v14 i also have no protection :'D

buuut... *loud thinking* if u can disable this test... the fan configuration "Target Temp" still available?
if yes... i can get the RPM data via API and can manage from my server via node-red a shut down (or turn off the power plug) ... if they drop under the limits i defiene... or whatever...

Either way, I see more possibilities than "never goes into operation" and yeah i like the firmware and webinterface :D

rdugan, 2024-08-16:
Yep, you can still use target temp as that's my addition on top of the stock fw.  And you shouldn't need to control the fans from an external source.  I'm assuming you just have them attached to the board headers?  My fan mode is just adjusting the PWM signal sent to that header.

Lyxon1337, 2024-08-17:
yes, that's right, I connected the fans to the board.
And apparently I don't fully understand the fan check :'D
I thought the fan check runs continuously and not just when booting.
But if your "target temp" still has control (and detects defects? - so that it doesn't have to be monitored externally via API) that sounds very exciting

well... I would like to test a firmware like that! :)

Lyxon1337, 2024-08-17:
okey now i tested a little bit more with your given informations.

Test setup - Like Picture :D
1. update again to pbv080_ks0ultra
2. shut down
3. plugin the old little fans
4. boot up and wait for fan test
5. setup "Fixed speed" to "0%"
6. change the fans to the noctua
7. setup "target temp" to 70 
than it's work without problems :'D

but if i unplug the fans it's still work... so yes... 
In the end, the fans are only checked once when booting.
There is apparently no "there is 0RPM but there needs to be more because setup is min >0%".
So I have to check the RPM externally via API anyway.

Or does the fan have to be "broken/unplugged" for a certain amount of time?

[image]

Brennerlein, 2024-09-10:
> @Lyxon1337 find me on Kaspa discord if you want to test a build w/ the fan check removed

Gibt es inzwischen eine Firmware ohne FAN check?

Lyxon1337, 2024-09-10:
yes and it's works :)
He gave me the firmware for testing (i write him in Kaspa discord (like he asked for)
no idea when this firmware will go live

Brennerlein, 2024-09-10:
I got no answer yet. Please can you send me this firmware too?