rdugan/iceriver-oc #73: Fan control reverts to asic setting on power loss

1Knightwing, 2024-08-16:
I generally run my PB pros on 100% fixed fan as my OCs run close to the 125* edge. When power is interrupted, they reset to the asic control setting, thus lowering rpm’s and dangerously increasing power board temps. The miner control panel still displays that they are fixed and 100% tho not the case.

rdugan, 2024-08-16:
I can't control what IceRiver does in the background.  If they decide to randomly switch fan modes, my fw can't do anything about it, unless it's constantly checking the speed every X seconds.  Which is exactly what my temp target fan mode does - so just use that.

closed 2024-08-16 by rdugan

1Knightwing, 2024-08-16:
Thanks! Thought about that but since there are 3 separate modes, fixed, temp, and asic, I assumed it would restart on asic with either. Will give it a try!Sent from my iPhone

rdugan, 2024-08-16:
It's always restarting on auto.  My firmware reapplies your settings a few minutes after startup.  But IR can always override that again later, unless i have a loop constantly reapplying, like in the target temp mode.

1Knightwing, 2024-08-16:
Got it! Prob doesn’t matter what the target temp is, long as min is 100%?Sent from my iPhone