rdugan/iceriver-oc #75: iceriver ultra

dropje20, 2024-08-19:
I put the firmware on it, now I can't log in password error 12345678, what's wrong

rdugan, 2024-08-26:
Is this still an issue?  Someone else recently had a similar experience, and it appears he was hacked.

Silent8CH, 2024-08-27:
I've found the password set in the OC firmware persists across updates and factory resets.

rdugan, 2024-08-27:
@Silent8CH yes, it does.  there's not really a way for me to reset the password during a hardware reset, as that is all handled in a complex routine that is part of the compiled IceRiver miner application, and my firmware uses a completely new database file it knows nothing about.

but my firmware gives you ssh access on ultras/ks5s now (and you can set it up yourself on the previous models), which means in the situation of a compromised system or lost password, you can simply delete the database via ssh.

Silent8CH, 2024-08-27:
How can I SSH to my Ultras? I only see ports 80, 443 and 4111 open to them.

rdugan, 2024-08-27:
port is 54188
credentials added by my fw are iceriver : 12345678

Strange2888, 2024-08-28:
> 
> but my firmware gives you ssh access on ultras/ks5s now (and you can set it up yourself on the previous models), which means in the situation of a compromised system or lost password, you can simply delete the database via ssh.
> 

@rdugan 
Can you tell me how to delete the database via ssh?

My KS0 Ultra was probably hacked this afternoon. Unfortunately, setting the miner to factory settings didn't help. With the original firmware I have access to the WEB GUI. As soon as I install your OC firmware I get "Login incorrect" when I log in.

Thank you for your help.

rdugan, 2024-08-28:
@Strange2888 Just delete /var/order/db

You could also try changing the password via the api, though if you were really hacked, they likely changed that key as well.

Which also brings up the point: if you're exposing the machine by forwarding 443 or something else, make sure you not only change the default password, but delete or replace the default api key as well

Strange2888, 2024-08-31:
Thank you, @rdugan 👍

it worked and I have access to my miner again.

I immediately changed the password and deleted the API so that it doesn't happen again.

rdugan, 2024-09-16:
I don't add accounts on earlier models.  You can follow [this](https://github.com/incidrthreat/iceriver_oc/blob/main/ECLIPSE.md) for adding your own access.

yesuaambros, 2025-06-09:
@rdugan I'm in the same situation as the rest of my colleagues and I don't have access to the web interface. I've reset the device several times and set the password to 12345678. I can't access it.

In my case, it's a KS0 Ultra.

I try to access via SSH on port: 54188.

I connect, but the same error occurs:
User: admin
Password: 12345678

And I get an "Access denied" error.

I updated the firmware again using the iceriver tools application.

If I install the original firmware, the website doesn't open. If I install the rdugan firmware again, the website loads, but I get a credential error.

According to your messages, I've tried using the username: iceriver and password 12345678 via SSH port 54188. And the same access error occurs.

I tried using Eclipse using the link you show, but Eclipse doesn't connect. (Would it also be port 54188??)