rdugan/iceriver-oc #78: API use from PHP

mactoolz, 2024-08-29:
Hello, 

i use PHP to use the API interface. Every time iam at the HTTP ErrorCode 404.

Do you have some tipps to solve my problem.

That is my API Request as curl config.

`    $ch = curl_init();
    curl_setopt($ch, CURLOPT_URL, $url);
    curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
    curl_setopt($ch, CURLOPT_SSL_VERIFYPEER, false); // Bei selbstsignierten Zertifikaten
    curl_setopt($ch, CURLOPT_FOLLOWLOCATION, true); // Folgt Umleitungen automatisch
    curl_setopt($ch, CURLOPT_HTTPHEADER, array(
        'Authorization: Bearer ' . API_TOKEN // API-Token für die Authentifizierung
    ));`

rdugan, 2024-08-29:
I would need to see the URL :)

mactoolz, 2024-08-29:
This is my URL oc curl ...

https://172.16.210.8:443/api/v1/overview

rdugan, 2024-08-29:
there's no /v1 in the endpoints

mactoolz, 2024-08-29:
ok, and what is the authentication mode ?

rdugan, 2024-08-30:
you've already included the authentication in the HTTP Authorization header

closed 2024-08-30 by rdugan

mactoolz, 2024-08-30:
`/**
 * 0003 - FN_ApiGetRequest
 * Führt eine GET-Anfrage an die API durch.
 * 
 * @param string $endpoint - Der API-Endpunkt, der angefragt werden soll.
 * @return array - Die Antwort der API als Array.
 */
function FN_ApiGetRequest($endpoint) {
    $url = API_ENDPOINT . $endpoint;
    
    print_r($url);

    $ch = curl_init();
    curl_setopt($ch, CURLOPT_URL, $url);
    curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
    curl_setopt($ch, CURLOPT_SSL_VERIFYPEER, false); // Bei selbstsignierten Zertifikaten
    curl_setopt($ch, CURLOPT_FOLLOWLOCATION, true); // Folgt Umleitungen automatisch

    // Authentifizierungsmethode auswählen
    switch (API_AUTH_METHOD) {
        case 'BASIC':
            curl_setopt($ch, CURLOPT_USERPWD, API_USERNAME . ":" . API_PASSWORD);
            break;
        case 'BEARER':
            curl_setopt($ch, CURLOPT_HTTPHEADER, array("Authorization: Bearer " . API_TOKEN));
            break;
        case 'DIGEST':
            curl_setopt($ch, CURLOPT_HTTPAUTH, CURLAUTH_DIGEST);
            curl_setopt($ch, CURLOPT_USERPWD, API_USERNAME . ":" . API_PASSWORD);
            break;
        case 'OAUTH':
            curl_setopt($ch, CURLOPT_HTTPHEADER, array("Authorization: OAuth " . API_TOKEN));
            break;
        default:
            return array("error" => true, "message" => "Ungültige Authentifizierungsmethode");
    }

    $output = curl_exec($ch);

    if (curl_errno($ch)) {
        $error_msg = curl_error($ch);
        curl_close($ch);
        return array("error" => true, "message" => "CURL Fehler: " . $error_msg);
    }

    $httpCode = curl_getinfo($ch, CURLINFO_HTTP_CODE);
    curl_close($ch);

    if ($httpCode != 200) {
        return array("error" => true, "message" => "HTTP Fehler: Code " . $httpCode, "response" => $output);
    }

    $response = json_decode($output, true);
    if (json_last_error() !== JSON_ERROR_NONE) {
        return array("error" => true, "message" => "JSON Fehler: " . json_last_error_msg());
    }

    return $response;
}`

mactoolz, 2024-08-31:
and now ????

rdugan, 2024-08-31:
404 means the url is wrong, as i explained above.  your auth header in the original post looks correct, which i've also already stated.  i can't debug your php for you.

I'm not sure what your question is.  here's an example curl request:

curl -k -H "Authorization: Bearer 5b281acc-de86-41bb-b14d-e266d9c9edbd" https://ks0ultra-0/api/overview

mactoolz, 2024-08-31:
Ok, thank you, i test it again.

I see it is a Bearer token, not only Token that is importent.

mactoolz, 2024-09-03:
it is not possible for me to connect from PHP to our firmware. But if use curl at the windows console, it is working !!!

mactoolz, 2024-09-03:
now it works ... i have set some curl options, but i dont know what was wrong. i test it ...

`// ###########################################################################################################################################
/**
 * 0003 - FN_ApiGetRequest
 * Führt eine GET-Anfrage an die API durch.
 * 
 * @param string $endpoint - Der API-Endpunkt, der angefragt werden soll.
 * @return array - Die Antwort der API als Array.
 */
function FN_ApiGetRequest($endpoint) {
    $url = API_ENDPOINT . $endpoint;

    print_r($url . PHP_EOL);

    $headers = 
    [
        "Authorization: Bearer " . API_TOKEN,
        "Content-Type: application/json"
    ];

    $ch = curl_init();
    curl_setopt($ch, CURLOPT_URL, $url);
    curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
    curl_setopt($ch, CURLOPT_HTTPHEADER, $headers);
    curl_setopt($ch, CURLOPT_CONNECTTIMEOUT, 10);
    curl_setopt($ch, CURLOPT_FOLLOWLOCATION, true);
    curl_setopt($ch, CURLOPT_SSL_VERIFYPEER, false);
    curl_setopt($ch, CURLOPT_SSL_VERIFYHOST, false);    
    curl_setopt($ch, CURLOPT_TIMEOUT, 30);

    // Authentifizierungsmethode auswählen
    switch (API_AUTH_METHOD) {
        case 'BASIC':
            curl_setopt($ch, CURLOPT_USERPWD, API_USERNAME . ":" . API_PASSWORD);
            break;
        case 'BEARER':
            curl_setopt($ch, CURLOPT_HTTPHEADER, $headers);            
            break;
        case 'DIGEST':
            curl_setopt($ch, CURLOPT_HTTPAUTH, CURLAUTH_DIGEST);
            curl_setopt($ch, CURLOPT_USERPWD, API_USERNAME . ":" . API_PASSWORD);
            break;
        case 'OAUTH':
            curl_setopt($ch, CURLOPT_HTTPHEADER, array("Authorization: OAuth " . API_TOKEN));
            break;
        default:
            return array("error" => true, "message" => "Ungültige Authentifizierungsmethode");
    }

    $output = curl_exec($ch);

    if (curl_errno($ch)) {
        $error_msg = curl_error($ch);
        curl_close($ch);
        return array("error" => true, "message" => "CURL Fehler: " . $error_msg);
    }

    $httpCode = curl_getinfo($ch, CURLINFO_HTTP_CODE);
    curl_close($ch);

    if ($httpCode == 401) {
        return array("error" => true, "message" => "HTTP Fehler: Code 401 - Unauthorized. Bitte überprüfen Sie die Authentifizierungsdaten.");
    } elseif ($httpCode != 200) {
        return array("error" => true, "message" => "HTTP Fehler: Code " . $httpCode, "response" => $output);
    }

    $response = json_decode($output, true);
    if (json_last_error() !== JSON_ERROR_NONE) {
        return array("error" => true, "message" => "JSON Fehler: " . json_last_error_msg());
    }

    return $response;
}`

mactoolz, 2025-04-12:
Hello, i want to test again the API to the miner.

For some few weeks i have PHP code it has works, but right now i have problems to get some states from the API.

Is the api the same or do you have change something ?

greetings