rdugan/iceriver-oc #79: ks0 ultra no longer connects to the pools after internet reconnect

Strange2888, 2024-08-31:
Hello 👋
After reconnecting my internet connection (e.g. after a DSL signal failure) my KS0 Ultra no longer connects to the configured pools. It tries the configured pools one after the other, but without success. 
After rebooting the miner, it works straight away and the miner is back online.

Does anyone else have this problem or does anyone know a solution?

rdugan, 2024-08-31:
Have you configured your router as suggested here?

https://github.com/rdugan/iceriver-oc#known-issues

BahmaNvidia, 2024-09-01:
Hi I have the exact same problem with my KS5L and KS1
KS5L on version 0.8.1 and KS1 on version 0.6.0
and I have done all of the suggestion above. but still when my DSL reconnects to the internet for any reason the miners won't connect to the pool until I restart them.

Strange2888, 2024-09-01:
> Have you configured your router as suggested here?
> 
> https://github.com/rdugan/iceriver-oc#known-issues

yes, i've done it, but the problem stays :(

rdugan, 2024-09-02:
Has nothing to do w/ TLS, which is only for the web interface, and wasn't even used prior to my v0.8.0 fw.

This is some artifact of IceRivers screwy network setup.  I'll have to dig into it to see if it's something i can address.

dm807cam, 2025-05-06:
Any update on this? I also experienced the same issue. Miners are not connecting to any pools after internet outage.