rdugan/iceriver-oc #81: Clock and Voltage offset

oberhammerd, 2024-09-04:
I know that every miner works better with different voltages and clocks, but just as a guideline. Which clock and which voltage do you use to operate the miner? i only use a 180W power supply and an external 5V fan

Spindler360, 2024-09-09:
Atm I run my KS0 Ultra at ~210 W including 12V external AC fan. 
[image]
Without modifications and better cooling (especially the power stages), I would not run a too high overclocking.  
My offset depends on the ambient temperature. I'm hoping for a long, cold winter so I can tease out a few more Mhs. ;-)

pungus4you, 2024-09-15:
HI, thank you for this beautiful firmware, I have ice river ks0 ultra  ..I installed this firmware but i made stupid mistake and now its showing me  chip voltage 0 .. I cant get in to change voltage option so I can change  when ever i go web page instant goes request time out and throws me out so I have to log in again .. i can still mine but low hash rate.. could some one help me with this please

rdugan, 2024-09-16:
@pungus4you no settings are applied for the first 2 minutes after reboot.  If hte problem is because you set too high voltage, then you should still be able to log in during that 2 minute window.  Otherwise a factory reset should clear it.

pungus4you, 2024-09-20:
Hi,  I have done just as you guided me . but its gives me same error .
when I  click miner option its says request time out and it takes me  log in page again, here I am posting some screenshot
[image]
[image]

pungus4you, 2024-09-20:
I just found . not sure but some how I made mistake ..  with voltage
is there anything can be done? thank you 
[image]

Spindler360, 2024-10-07:
Go to the “Miner” menu item and try this: [image]

[image]

Keep in mind, that I also have a 330 W power supply unit (250 - 280 would also be sufficient), 3D printed fan shroud with separate 4500 rpm fan. Don't forget the copper heatsinks on the power stages. ;-) 

If there are problems, reduce the clock and wattage by a few units until the miner is stable (e.g. clock offset 195 and voltage offset 180). 
About 600 MHz +/- should be possible depending on the silicon lottery and ambient temperature.

pungus4you, 2024-10-16:
Hi @Spindler360,

Thank you for your kind help.

I’m facing an issue where, whenever I click the "miner" option, I instantly receive an error message saying "request timeout," and it redirects me back to the login page. This happens regardless of whether I click the option immediately after the page loads or later—it still gives me the same error. Due to this, I’m unable to make any adjustments to the miner. 😔

rdugan, 2024-10-17:
@pungus4you ssh into the machine (iceriver / 12345678, port 54188) and delete the /var/order/db directory

`sudo rm -rf /var/order/db`

will include something in the next update that does this on a factory reset

pungus4you, 2024-10-17:
@rdugan Thank you, I will try that 🙏