rdugan/iceriver-oc #83: Fan Error after Wake via API

marchapunkt, 2024-09-10:
I am experiencing an issue with my KS0 Ultra. 
I am using a smart home system to put the Miner to Sleep, and wake it up at certain circumstances via HTTP Request, but as soon as the Miner wakes up, it identifies an issue with the FANs that results in a very low hash rate. 
After rebooting the miner everything looks ok and it runs as expected.

rdugan, 2024-09-10:
what fan mode are you using?

marchapunkt, 2024-09-18:
Just tested around. When using target temp it is not working, when using ASIC Controlled it is working.

rdugan, 2024-09-19:
There must be something more to it than this. I just tried on my ultra, and it works fine w/ target temp.  I'm guessing you have a fan going bad.