rdugan/iceriver-oc #85: KS1 on 0.8.1 OC Firmware won't let me change pools

JohPePe, 2024-09-14:
Hi!

Had my KS1 on Iceriver firmware, Installed PBFarmer ver 0.8.1, changed pools. Everything looked ok.
Then I saw, my top pool was a solo pool (solo.bugna...) so I changed it to bugna... and klicked save.
Then the problem started, suddenly I had two no 5 lines. So I tried a restart. It got worse.
Now every time I try to change pools (Miner/pool config), I get a 'Request timeout' and then I get kicked out to the login screen.

Even tried after reset to IR org firmare, and installation of your 0.8.1 firmware, the same problem, no change.
I suspect the pool-data got messed up, somehow.
Is it fixable?

Running org IceRiver firmware, and it's working fine, so the miner seems ok.

rdugan, 2024-09-16:
Have you tried simply adding a new pool, saving it, then deleting the old one?

JohPePe, 2024-09-17:
Yes, it did not work. Same problemHad to install the previous version 0.7.1 That version works just fine!/Johan16 sep.

JohPePe, 2024-09-17:
Hi!After resetting the KS1 to stock firmware, rebooting, reinstalling your 0.8.1 version, it booted but seemed broken. I could not change, or see any pools, displayed a timeout message. The I reverted to stock firmware again, and tried your 0.7.1 version. It worked just fine! /Johan16 sep.