rdugan/iceriver-oc #87: KS5L No overclock

Tonykostansek, 2024-09-16:
If i increase the Clock speed on my KS5L the hash rate actually drops.  Im assuming this is a Voltage thing. But not sure.  The most I have been able to do is a 10Mhz increase.  And that didnt really add any hash rate.   The new interface is very nice though.  It has everything the stock interface needed to have.  Any one getting any stabile hash increase with this firmware?

rdugan, 2024-09-16:
25mhz is pretty much the max.  The one i test on starts losing hashrate above 19mhz.  10mhz (which is likely only actually 6.25mhz) is basically a 1% hashrate increase - enough to cover the fee.  You might be able to do 13mhz.  But it's not at all surprising if you can't go above that.

loic38210, 2024-09-21:
hello I find myself in the same situation as you I can increase my blood pressure (MV) but it is not possible with the ks5l.think you have to add it with an update? I'm used to it and I won't want to put another meter just for that

loic38210, 2024-09-21:
or do you know a firmware that I could install instead thank you

loic38210, 2024-09-21:
I switched to water cooled KS5L

rdugan, 2024-09-21:
Voltage control is not possible

loic38210, 2024-09-21:
I've seen people manage to change the voltage on other software.otherwise what's the point of being able to change just MHz?

loic38210, 2024-09-21:
I've seen videos on youtube where the person paces them in water cooling and with more than 15t of power instead of 12? How does it do it?

loic38210, 2024-09-21:
I am French and understand English very badly, I am obliged to translate

rdugan, 2024-09-23:
> I've seen people manage to change the voltage on other software.otherwise what's the point of being able to change just MHz?

Not all clock changes require voltage changes.  Please read the documentation on the tuning process.

Regarding how other people run 15T, you'd have to ask them.  My guess is they've removed the stock fans, freeing up some additional power for the hash chips.

closed 2024-09-23 by rdugan

loic38210, 2024-09-23:
ok thank you for calf rapunzel may I read the documentation for the settings and precisely c writes to increase the clock and as soon as the power starts to no longer respond or decrease increase the mv?    but we don't have the MV?and we don't even earn 1t and as said in the other comments so do you know another firmware or place where I could inquire? Thank you

rdugan, 2024-09-23:
I already answered this.  Voltage control is not possible.

And it wouldn't matter even if it was - the PSU is already maxed out.

This is not an issue with my firmware.

loic38210, 2024-09-23:
ok sorry I have a hard time understanding I am new in minig.il may happen to lose power by just increasing the clock a little while my power was only 3100w so still 250w pk its decan in power ?here this nai which power in watt I can even consume me of 3000watt the mv represents the powerful the power in watt ?thank you for veal rapunzel