rdugan/iceriver-oc #90: Unable to retain underclocks/undervolts

miodraggolubovikj-allocate, 2024-10-07:
I have a defective KS0 Pro that only works on -125 CC offset and -40 voltage offset. But when set, after several minutes the clocks get incremented back to 550 by 25mhz. Why is that? Is it possible to retain the underclocks without returning to stock values?

rdugan, 2024-10-07:
my fw will only reset the tuning when putting the machine into sleep mode, or when the fee connection isn't working properly.  In both cases you'll have an indication in the UI.

I can't think of any mechanism the stock fw has for resetting tuning params, other than a complete restart of the miner, and even then, my additions will reapply the offsets.

rdugan, 2024-10-07:
Ok, thats not a reset.  Either there's a bug in my fw, or the stock fw is overriding the settings.  Are you using my latest release?

miodraggolubovikj-allocate, 2024-10-07:
it was behaving the same on the v80 and the newest version. I am currently using the latest version.

miodraggolubovikj-allocate, 2024-10-08:
@rdugan do you have any plans to address this maybe in your next release? I have 2 miners that are unusable on the stock clocks and bootloop crash all the time.

rdugan, 2024-10-09:
I was looking into it.  It is indeed the stock fw overriding the clock settings.  This would happen on a pro for any underclock, so shockingly it seems you're the first person to ever try it, lol.  Anyway I can give you a test build if you find me on discord.

miodraggolubovikj-allocate, 2024-10-13:
Hi @rdugan I was able to upload the custom firmware today but it misbehaves. I wanted to revert back to stock FW then start from scratch and upload the file from there, but every way i try to revert back(button press 20 sec) it never shows the UI but if i use the iceriver batch tool it recognizes the miner and i can upload your fw back to get the UI working. Is there a custom way to rollback to stock?

rdugan, 2024-10-14:
Check the last item here:
https://github.com/rdugan/iceriver-oc?tab=readme-ov-file#known-issues

nordmann666, 2024-10-31:
> I was looking into it. It is indeed the stock fw overriding the clock settings. This would happen on a pro for any underclock, so shockingly it seems you're the first person to ever try it, lol. Anyway I can give you a test build if you find me on discord.

i can find you on discord, but cant send u message