rdugan/iceriver-oc #91: ERR_CONNECTION_REFUSED after update

MiraMaX166, 2024-10-15:
I updated version 080 to 081 for ks0pro. After that I lost access to the device, the browser writes ERR_CONNECTION_REFUSED. I returned the firmware to default. Did a full reset. I tried to flash 081 again, but the result was the same - ERR_CONNECTION_REFUSED. I reset everything again and tried to flash 080, but now this firmware also gives ERR_CONNECTION_REFUSED. That is, now I can’t install any firmware other than the standard one or the firmware from Swift. What could be the matter? This is not an isolated case, all three of my devices are now suffering from this problem.

MiraMaX166, 2024-10-15:
Version 071 just installed without problems

rdugan, 2024-10-15:
081 uses https

https://github.com/rdugan/iceriver-oc#new-authauth-routines

https://github.com/rdugan/iceriver-oc#known-issues

UHeinz13, 2024-10-20:
I have 5 KS0 Ultra, and one of the 5 miners has exactly the same problem. If I want to log in, I get to the main page, but when I click on the "Miner" menu item, I always get the message "Connection refused" and I end up back at the login window. It works problem-free with the 4 other miners. Already tried everything, cache cleared, different browsers, from another PC, always the same behavior.

rdugan, 2024-10-21:
@UHeinz13 these are not the same issues.  Your's is from some change you made to your pool setup that corrupted the database some how.  Just ssh into the machine and delete the /var/order/db directory (iceriver / 12345678, port 54188)

UHeinz13, 2024-10-21:
I deleted the two files ir.bak and ir.mdb, restarted and now it works. Many many thanks

bowuesser8107, 2024-10-28:
> @UHeinz13 these do not sound like the same issue.  Your's is from some change you made to your pool setup that corrupted the database some how.  Just ssh into the machine and delete the /var/order/db directory (iceriver / 12345678, port 54188)

Thanx for this tip.... 👍

AndyBochum, 2024-12-17:
hello! is it the same by ks0 pro ? iceriver and 12345678

rdugan, 2024-12-17:
@AndyBochum no, pros have the TCF port open so people were able to add ssh accessible accounts on their own by going through this process: https://github.com/incidrthreat/iceriver_oc/blob/main/ECLIPSE.md