rdugan/iceriver-oc #93: KS1, KS2 unavailable after reset

MrKTOTO, 2024-11-14:
several devices have already frozen during operation, it writes an error that the device is not available, rebooting does not help, after resetting to factory settings on version 0.8 and higher, the equipment interface completely stops loading, resetting by the reset button does not help. However, those devices that were on version 0.7.1, resetting by the button helped. As a result, I already have 4 bricked devices due to the new firmware, everything I tried to restore does not help.

MrKTOTO, 2024-11-14:
You cannot use versions 0.8 and higher, this will brick your device.

rdugan, 2024-11-14:
You need to read the README.

https://github.com/rdugan/iceriver-oc#known-issues

closed 2024-11-14 by rdugan

rdugan, 2024-11-15:
It was almost certainly a cache/refresh problem before you reset, now it sounds like the last item. The only changes from 0.7 to 0.8 were replacing the web app (including migrating to https).  The mining side was unchanged, and factory reset has always been stock IR code.

Otherwise it's almost certainly your network setup - also all discussed in that section.

MrKTOTO, 2024-11-15:
Okay, well, how do you then explain that the devices don't work in three differently configured networks?

rdugan, 2024-11-15:
Everything i can explain / tell you is in the docs I linked.  I don't have any further info I can provide.

rdugan, 2024-11-15:
not before resetting.  after.

read the last item.  it's referring to what happens AFTER you reset.

MrKTOTO, 2024-11-17:
[image]

See, it's in incognito mode

MrKTOTO, 2024-11-17:
Device bricked, your instructions not working!

MrKTOTO, 2024-11-17:
this does not happen on versions below 0.8, only on 0.8 and above

MrKTOTO, 2024-11-17:
[image]
Another computer

MrKTOTO, 2024-11-17:
[image]

MrKTOTO, 2024-11-17:
[image]
From phone

MrKTOTO, 2024-11-17:
browser cache from all devices? do you take me for an idiot?

chad5150, 2024-11-17:
Did you reset the device through the web interface or on the physical device. I’ve reset my devices through the web interface and had weird issues both on stock firmware and this firmware.

rdugan, 2024-11-17:
> browser cache from all devices? do you take me for an idiot?

1. Being an ass isn't going to convince me to help you - especially with a problem with your stock firmware
2. Neither issue i've told you to focus on were cache - one is a browser bug reverting from https to http (and yes, it happens on all browsers you've used to view the https site), and the other is network configuration.  If you simply confirmed you've taken the steps I keep asking you about, this would move much more quickly.
3. Clearly your device is not 'bricked' as you keep saying, considering there is a web server running and answering requests.

Now, if you'd like my help, please answer the questions below:

1. Have you performed a hardware reset by holding the button down for at least 30 seconds?  Have you tried doing this a few times?
2. what happens when you type http://192.168.0.95/user/login