rdugan/iceriver-oc #94: "ASIC request failed" cant do anything

11ssA11, 2024-12-05:
it stuck on ASIC request failed I can log in to gui but i cant change anything. I somehow go back to stock firmware and tried again but still same. 
[image]

hacksr00t, 2024-12-05:
This happens to me a lot.. two KS3M in two specific use case.  One is I had a bad power supply and didn’t know it.  Power off then on and it says this right away and changing the power supply fixed it.

the second is every two days for me on both units with latest release.  I underclock (a lot..  like -300 on my clocks) my asics will run for two days at most..  and then they fail and this is the message I get when I login.  Reboot won’t work, however Power cycle and everything is perfect again.

hope this helps.  It would be cool if my KS3m’s lasted for more than two days but I figure it’s an odd use case for most lol

rdugan, 2024-12-11:
Did you try just force-refreshing (shift+refresh) the page?  You'll see this message when the session expires - reloading the page should redirect you to login

11ssA11, 2024-12-11:
> This happens to me a lot.. two KS3M in two specific use case. One is I had a bad power supply and didn’t know it. Power off then on and it says this right away and changing the power supply fixed [...]

Well it happens on default settings as well so i cant change any clock settings. nothing at all.
I do think my psu is fine cuz i am using swift right now and it is fine more than a week.

11ssA11, 2024-12-11:
> Did you try just force-refreshing (shift+refresh) the page? You'll see this message when the session expires - reloading the page should redirect you to login.
> 
> Otherwise, it means the miner died. I can't really tell you why that would happen - it's a compiled program from IceRiver - but it's probably because of inappropriate OC. My fw should restart the miner automatically when it dies, but that may not be working correctly on newer models, as IceRiver started deleting the application after starting it. It's something I'm addressing in the next release

Device is no dead. I can turn back it to default via iceriver app. But i cant use oc firmware. Yes i tried force refresh and incognito mode even 3 diffrent browser and 2 diffrent pc. 
And no i didnt overclock. I cant get that state yet. It just unstable at stock settings at oc firmware.
But it works totally fine on swift firmware . Any idea ?

rdugan, 2024-12-11:
When I said miner died, i mean the miner application, not the device.  I have no idea why it would only die on my fw if you've never changed clocks/voltages.  Unless it's a problem with IRs latest fw update, which my fw is built on top of.  

Which stock version are you running?

11ssA11, 2024-12-12:
> When I said miner died, i mean the miner application, not the device. I have no idea why it would only die on my fw if you've never changed clocks/voltages. Unless it's a problem with IRs latest fw update, which my fw is built on top of.
> 
> Which stock version are you running?

Should be latest  : ICM168_3_2_15_ksupdate.bgz

rdugan, 2024-12-17:
What does it say at the bottom when you're running stock?  Also, what device are we talking about?