rdugan/iceriver-oc #95: KS0 Pro on 0.8.1 OC Firmware won't let me change pools

AndyBochum, 2024-12-17:
Hallo!
Ich habe einen zusätzlichen Pool in die Liste eingetragen. Jetzt komme ich nicht mehr in die Minerliste um etwas zu ändern. 
Auf der Home Seite steht als Pool

stratum+tcp://de.kaspa.pool.pbfarmer.cc:1208 | kaspa:qrhc5t3pysv95h6dwx5jtax8myy5dms7nqwq8mspuxmfm436k3326jxkdspym.irproxydefault

Ich vermute das es an der Liste liegt. Anzahl der Pools vieleicht zu geworden hoch? Ich hatte einige male etwas an den pool geändert, gelöscht und neu erstellt. Die Pool ID hatte sich dann erhöht. Ein zurücksetzen auf werkseinstellungen geht. Danach erneut update auf z.b. version 0.8.1 bis 0.7.1 leider ohne erfolg. 
Vieleicht ist auch der bootloader defekt? Oder sogar falsch?

Version BOOT_2_0.BIN | image_1.0 | pbv081_ks0pro_miner | pbv081_bg
IP 192.168.178.64 MAC 00:0a:32:00:41:9d

AndyBochum, 2024-12-17:
Hello!
I have added an additional pool to the list. Now I can no longer access the miner list to change anything.
On the home page the pool is

stratum+tcp://de.kaspa.pool.pbfarmer.cc:1208 | kaspa:qrhc5t3pysv95h6dwx5jtax8myy5dms7nqwq8mspuxmfm436k3326jxkdspym.irproxydefault

I suspect it is because of the list. The number of pools has perhaps become too high? I had changed something in the pool a few times, deleted it and created a new one. The pool ID had then increased. Resetting to factory settings works. Then I updated again to e.g. version 0.8.1 to 0.7.1, unfortunately without success.
Perhaps the bootloader is also defective? Or even wrong?

Version BOOT_2_0.BIN | image_1.0 | pbv081_ks0pro_miner | pbv081_bg
IP 192.168.178.64 MAC 00:0a:32:00:41:9d

rdugan, 2024-12-17:
There's an issue w/ the way some people are adding pools, which I've been unable to re-create myself.  It causes some database inconsistencies.  And unfortunately in that release, the database is not deleted when you do a factory reset, so the only way to recover is to ssh into the machine and delete the /var/order/db directory.

However, I have a new pre-release version which will delete the DB on factory reset, so if you find me in Discord, i can give you access to that.  You would just need to install the new version, then factory reset, then re-install.

Anfaenger2022, 2024-12-19:
Big thank to You. It is working
Fine

fatjerem, 2025-01-15:
hi @rdugan ,
i have the same problem with 9 pools with no ssh possibilities :(
can you send me your pre release version to clean db ? 
thanks a lot

rdugan, 2025-01-16:
@fatjerem you'll need to send me a FR in discord.  Alternatively, you can set up ssh access yourself following this:

https://github.com/incidrthreat/iceriver_oc/blob/main/ECLIPSE.md

fatjerem, 2025-01-21:
Hi, it's work for me on KS0 pro, thank you a lot @rdugan !!!! :)

BahmaNvidia, 2025-05-14:
Hi 
I have a KS5L with v0.8.1 and my pool configuration is messed up too and I need the pre-release firmware you have mentioned to clear the DB
I have mentioned you in discord but can't send a FR to you there.
please add me my username on discord is slicky3087
thank you